mclear/Android_NFC_Ring_Unlock #120: Fix app

pawel37, 2016-05-01:
Hello can u make the app not lock when I am on a call and after I put the phone to my ear when in call the app locks I don't like that at all psl  fix it

JohnMcLear, 2016-05-01:
Can you please provide steps to replicate this problem?

pawel37, 2016-05-02:
Call make a call put phone to ur ear and then try to look up number when in
call u have to unlock. Screen turn and app locks phone it's should not lock
phone in a call.

JohnMcLear, 2016-05-05:
I don't understand this, can you be more coherent please?  Please use proper English!

TJ-G, 2016-05-05:
I don't think this behavior has anything to do with the NFCRing app. It's standard, customizable android behavior to have the screen turn off when the proximity sensor detects your phone next to your face, and to lock after X amount of screen off time. You can either turn off the proximity sensor feature, and let your screen stay on during a call, or change the amount of standby time until the screen locks.

If you're just annoyed that you have to unlock the phone to end a call, many android phones have the ability to let the power button end a call. That doesn't require you to unlock the phone, and would likely solve the root of your problem. If you have questions, I suggest you google 'android screen lock during phone call'.